wagiminator/C64-Collection #1: Epyx Freeload wrong LED voltage ?

il-cyclone, 2021-11-04:
Hi,
Ordered a bunch of Epyx Freeload PCBs from Jlcpcb, and have now assembled two of them.
Both work with the exception of the LED which would not illuminate on either one. When I measure the voltage over the LED it is 5V instead of the expected 1.8-1.9V. The resistor used is a 470R as specified in the BOM.

It feels like the LED is fed 5V without going through the resistor. My knowledge of electronic design is pretty bad, but when I look at the PCB, only one end of the 470R resistor is connected to the circuit, is this correct ?

[image]

il-cyclone, 2021-11-04:
Update:
Build a third one, and now the LED is lit with around 1.8V over the pins. However, it is constantly lit when the cartridge is idling. When I start loading from disk, the LED shuts off, and comes back on when the loading is complete. I was kind of expecting the opposite behaviour.
My only deviation from the BOM is that I use a 27C512 eprom instead of a 27C64, but they seem to have the same pinout so I guess that should not matter.

wagiminator, 2021-12-11:
Hi, I don't know why the LEDs don't work on your first boards. It cannot actually be due to the EEPROM. I rather suspect that the LEDs are defective, in this case you would actually measure 5V. You can check them with the LED test function of a multimeter.
It is intended that the LED always lights up and goes out during operation (Apple would say: it's not a bug, it's a feature).