wagiminator/C64-Collection #4: C64 TapeBuddy64

Jaroo-z, 2022-07-24:
Hello,
Nowadays, it is very difficult to buy microprocessors used in this project.
Is it possible to load the program into the microcontroller with 2kB of memory? For example, ATTINY 214, 204?
Or rewriting the program to Arduino Nano?

Thanks in advance.
Jarek

wagiminator, 2022-11-24:
The problem currently exists with almost all controllers. The ATtinys are occasionally available on Digikey. Of course, the program can also be rewritten for other controllers, but I don't have the time for that at the moment.

Aisflou, 2022-12-12:
Can the code run well in the 10mhz variant? ATMEGA644PV-10AU its pin compatible and seems to be found sometimes at reasonable prices...