wagiminator/C64-Collection #8: C64 TapeBuddy64 for theC64 Maxi

Smiril, 2023-06-03:
hello I got a question about bring tape to theC64-Maxi (retro console) ....

because I got a bunch of tapes and a datasette but no part between datasette and USB

where can I check the full assembled plug? kickstarter?or some other funding site for these people who want a part of this spice?
hope this will be answered

THX

Aisflou, 2023-06-04:
Te real cuestion is, can you execute python in the C64 Maxi? if yes, then you just need to orther the pcbs from the gerbers provided, order the parts from the BOM provided, wait for all to arrive, solder all toguether, program the microcontroller, and you are done! :D just connect your datasette to the C64 Maxi via your newly made adapter execute the python scripts and enjoy! Dont know if anybody makes this already available for purchasing, maybe searching in ebay show something.

Smiril, 2023-06-04:
theC64-Maxi  can run 32bit arm binaries from C/C++ ... but new firmware must provided ... ill read the LICENSE ...
so on this PCB I mean can build ... for private or commercial with dump LICENSE and COPYRIGHT holder if sold and no limitations could be made ... python ?!? ... I do not know ...  it's a linux subsystem with vice-2.4 emulator on a dual core with 8MB nand flash storage for system ... possible to add usb-stick to load software for vice and firmware

here is the firmware:
```
git clone https://github.com/retro-games-ltd/redquark-sixtyfour.git
cd redquark-sixtyfour
cd src
git clone https://github.com/retro-games-ltd/vice-2.4.git vice2
sudo apt install pkg-config libasound2-dev libjpeg-dev libpng*-dev bison flex g++
cd ..
./configure
make
```

Aisflou, 2023-06-04:
The boards and software allow to make, sell or whatever you want no problem, only attributtion must be done. For the software part you must do your research to try and adapt  the software to your needs, the provided software here works with windows, linux, mac, as soon as it runs python, you need to find if you can run python in the C64 Maxi... and if you can install drivers for the serial to usb converter... the info in the page states:

"Python needs to be installed on your PC in order to use the software. Most Linux distributions already include this. Windows users can follow these [instructions](https://www.pythontutorial.net/getting-started/install-python/). In addition PySerial and Tkinter (8.6 or newer) must be installed. However, these are already included in most Python installations.

Windows users may also need to install a [driver](http://www.wch.cn/download/CH341SER_ZIP.html) for the CH340N USB to serial adapter. This is not necessary for linux users."

So thats it, if you can run python and can install some ch340 driver then in theory it would work in the C64 Maxi.

wagiminator, 2023-06-23:
I don't know what you want to achieve in the end. If you only want to use the programs/games on your old cassettes on the C64maxi, then it is sufficient if you convert them into .tap files using a normal PC and the adapter, then save them on a USB stick and insert it into the C64maxi. But if you want to load games on your C64maxi directly from the Datasette, then it probably won't work that easily.