wagiminator/C64-Collection #9: USB-C to 1541-II Power Adapter Case Modification Request

TankedThomas, 2023-06-20:
Hi Stefan,

Just wondering if you could please modify the STLs for 3D printing the 1541-II power adapter case.

I tried to get it printed through JLCPCB (don't currently have my own printer) and they said the walls are too thin (1mm) for any material and therefore risk breakage, so they recommend 1.5mm thickness. 
I'm clueless about 3D modelling so I really don't have any idea on how to modify the models myself.

Also, it was mentioned in #7 that the bigger footprint for L1 makes the board a bit too big for the case, apparently? I'm not sure about this as I haven't tried it myself, but if necessary, can you please modify the case for this too?

Hopefully this isn't too big a request.

wagiminator, 2023-06-23:
Puh, let's see if I can find the time to do it.

TankedThomas, 2023-06-27:
Thanks, that would be really wonderful of you if you could do so! I tried to figure it out myself but couldn't, sorry.