wagiminator/C64-Collection #11: DumpMaster64 - Unable to burn bootloader

NewGuy0101011, 2023-07-27:
Hello,

Firstly I want to say thanks for all of your work. I have built several of your projects now and this has been my first introduction to building, programing and getting a working thing at the end of it all. Super rewarding and super fun!

I have recently got 5 DumpMaster 64 from a friend that I am trying to program to get to work. These came to me preassembled and everything looks ok. I plug the unit in and the power led illuminates as expected. When trying to program the device using Arduino IDE and the instructions given on your page, i continually get an error. I have downloaded the MegaTinyCore package, set all of the chip, clock and programmer settings as suggested but when i click burn bootloader, i get an error message everytime.
[New Text Document (3).txt](https://github.com/wagiminator/C64-Collection/files/12178588/New.Text.Document.3.txt)

I have tried all of the 5 boards i have ensuring that the switch is set to UDPI but the same result each time. I have attached a text document that shows the error code i keep getting.

Thanks in advance for any help.

wagiminator, 2023-07-28:
Did you choose the correct COM port (Tools -> Port) ? Have you tried uploading the firmware via dumpmaster-gui.py? If yes, what error message do you get?

jcthe2nd, 2023-08-06:
Hi 

I am trying to program this awesome project you created thank you. I am able to burn bootloader but im unable to  flash the firmware via Arduino IDE when i click upload i get this msg. 
"Sketch uses 3547 bytes (43%) of program storage space. Maximum is 8192 bytes.
Global variables use 336 bytes (65%) of dynamic memory, leaving 176 bytes for local variables. Maximum is 512 bytes.
A programmer is required to upload"

I get the GUI python script to run and see the GUI but nothing seems to do anything when i click flash firmware.

Do I need the firmware.bin file in the same directory? 

All switches are set correctly and COM ports and all other settings.

wagiminator, 2023-08-06:
Did you install PySerial?

jcthe2nd, 2023-08-06:
thank you it works perfect in linux but a little trouble in windows thank you

wagiminator, 2023-08-08:
Great! However, it should work with Windows as well (I've tested it on a Windows 10 PC).