wagiminator/C64-Collection #15: 1541-II Power-board - question: USB data through?

rmpel, 2023-12-22:
First of all; AMAZING! I have been toying with this idea but lack any electronics skills to design a board like this. I never thought to use it for a 1541-II (or any other with internal PSU), I "need" one for a ATAPI ZIP100 to USB conversion I recently finished (with a big 220v PSU). I would really want a single cable solution; for that I would need the USB data pins (USB2, simple 2 wires).

Here is the question; Does this board design support just tapping into the two data wires on the connector? I noticed in the diagram the pins are connected to one of the ICs, really no idea how all this* works ;)

Thank you for your amazing board, if I can find where, I will order some (would start with 4, 1 for my 1541, 1 for my 1571 if I ever feel confident enough to dremel the case, one for my ZIP drive conversion and one space because I will most definitely f*** it up!.

*) Power delivery is magic!

wagiminator, 2023-12-27:
Hi, the data lines on this board are not routed to the outside as it is not necessary for the intended purpose. The CH224K is of course connected via the data lines, as this is where the charging protocol runs and the output voltage is negotiated.

rmpel, 2024-01-07:
Sorry for the late response, didn't get notifications?? will look into that.
Thank you for your response, I saw indeed that the data lines are not exposed, but, could I, in theory, solder the two data-wires of a USB cable directly on U1, pins 4 and 5 and connect that to my USB to ATAPI adapter?
As I said, Power Delivery is magic to me. 
If these wires were unused, I would not hesitate, but as they go to the PD chip for negotiation, can I just hook up an extra device on the same wires?? 

When I eventually get the boards (PCBWay with assembly takes a while ;) ) I will just try with a disposable USB device, in case It blows ;)

Again, thank you for your response and best wishes for 2024.

rmpel, 2024-01-07:
I may have found an answer in the datasheet for the CH224K;

> If you don't need to use A port protocol (various protocols realized by DP, DM communication), you can choose CH221K model. If you want to shield these protocols on CH224K/CH224D, you need to disconnect the DP/DM pin of CH224K/CH224D from the Type-C interface.

So I guess just adding the wires will not work, but I think I can manage. I ordered enough for me to potentially destroy one ;)

wagiminator, 2024-01-07:
If you only want to use USB-PD as a charging protocol, then that shouldn't be a problem overall. USB-PD only uses the CC lines, you no longer have to connect D+ and D- to the CH224K and you can then route them out for your needs. Take a look at the data sheet, I think on the CH224K you have to connect the pins for D+ and D- together in this case.

rmpel, 2024-01-16:
I had planned to do just that, the specs say to short the D- and D+ so I was planning to route a single "input" USB-C to split into Power+CC to the board and D-/D+ and power from the 5v output of the board to the USB device, however, things didn't go as planned. I'm creating a new issue for that, but I do not expect any support, though do hope for some input. Things will be clear in that issue. 

Thank you for all your time so far, and if I ever get it working, I will return here.