wagiminator/C64-Collection #16: "copy prg to disk" software feature request

COREi64, 2024-01-13:
Hello.  The Diskbuddy 64 is awesome.  

I was wondering if a feature could be added to the software

Copy PRG to disk

I know you have copy PRG from disk but it would be great to go the other way.  I have some PRG files that I'd like to copy to disk with this solution.  I'm sure I could find some other software that could do it for me, but seems like this would be really useful and handy to do right from the diskbuddy64 gui.

Thank-you

wagiminator, 2024-01-22:
Hi, yes, that is basically a good idea. However, I'm busy with a number of other projects at the moment, so I won't be able to get to it any time soon. But I'm putting it on my (far too long) to-do list.

COREi64, 2024-01-24:
Thanks for the consideration.  Hopefully it's not too many years down the road, but of course I understand how other projects would come first.  Great device and software BTW, enjoying using it.