wagiminator/C64-Collection #17: Boards not working - don't know how to proceed.

rmpel, 2024-01-16:
I have been thinking if I should report/ask here or not and decided that it could be beneficial for others, so here goes.

I have had PCBWay build 5 boards for me, including assembly, as my eyesight does not allow me to solder SMD (that and the lack of equipment ;) ) (This wasn't an easy task as there is one part missing in the data, which is, if I'm not mistaken, the pick-and-place index?? In the end, they went of pictures and information I could determine from your schematics)

The boards look fine, all components are mounted correctly, but when connecting to a (verified PD 12 volt compatible) power supply, all I get is 5 volt on both outputs, and the "stable power" LED does not come on. (No shorts, that's great ;) but unfortunately, no 12 volts either)

I had given up, but maybe with a few pointers I could salvage the modules and proceed with the projects, and other people that run into the same issue can read a solution here, if I/we ever find one.

As said in the other topic; I do not expect your support, it's open-source, it was my own choice to have them built, you don't owe me anything ;)
That said, any and all information you are willing and able to provide will be greatly appreciated.

Remon.

p.s., I have a ugreen 100watt PD supply at work I'm taking home today, it literally says 12v 3A as one of the output modes on the unit, so if that doesn't work, then issue confirmed ;)

rmpel, 2024-01-24:
Hi. Once again, thank you for your response. I finally had a chance to test with the Ugreen 100Watt PD PSU, and ... it works fine. So now the question is; why would an Apple MBP PSU not support 12 volts but does 5 volts, 19 volts and 24 volts ...
And why would a laptop not provide 12 volts by PD protocol... 
Anyway, all boards produce 12 and 5 perfectly. 

The plan goes to the bin, but at least it wasn't money wasted ;)

wagiminator, 2024-01-28:
USB-PD is not always a simple story: source, sink and the cable in between must work together.