wagiminator/C64-Collection #18: DiskBuddy/Master Supported Drives

SteveGuidi, 2024-03-10:
Will the DiskBuddy and DiskMaster work with other IEC drives like the 1571 or 1581?  It seems like the XUM devices are more suited for that task and was generally looking for an alternative.

briskspirit, 2024-03-10:
Also curious about supported drives.

wagiminator, 2024-03-10:
I'm not sure since I only have a 1541.

briskspirit, 2024-03-10:
But do you have 1541 or 1541-II? Ordered parts to make a board, will test with single sided 1541 and double sided 1571

wagiminator, 2024-03-10:
I have both a 1541 and several 1541-II. However, they are fully compatible with each other anyway. This is probably different with the 1571 and 1581, so there might be issues here.

briskspirit, 2024-03-10:
Isn’t one is single sided and another is double sided? Anyways will try soon-ish

SteveGuidi, 2024-03-10:
> Isn’t one is single sided and another is double sided? Anyways will try soon-ish

Yes, the 1571 has double-sided operations, but only when you put in that mode.  The default mode of operation is just like a 1541.

I suspect the 1581 will work as well, as long as you don't use operations that require things like partitions or extra tracks (formatting).  I haven't looked at the source code for interfacing with the IEC bus yet, but I suspect we'll see these details there.

wagiminator, 2024-03-11:
If the 1571 has a 1541 compatibility mode then it will probably work. But things look significantly different with the 1581. The problem is that for speed reasons I didn't use the standard Kernal routines but instead developed my own fast loading system, parts of which also runs on the drive.

briskspirit, 2024-04-06:
1571 confirmed to work, thanks for great tool!

wagiminator, 2024-04-14:
Great, thanks for testing it out!