wagiminator/C64-Collection #21: Tapebuddy64: tape-gui.py throws tkinter error, doesnt show read data visualisation

InsaneDruid, 2024-09-05:
Tested with Python 3.12.5/tkinter 8.6 on Windows11 and Python 3.10.12/tkinter 8.6 on KDE NEON
After clicking read from tape to tap file and selecting the output file, only a blank window appears with no graphical output, while tkinter throws the following error

_Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\tkinter\__init__.py", line 1968, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "D:\TapeBuddy64\tape-gui.py", line 174, in tapeRead
    canvas.create_rectangle(0, 0, 256, 128, bg='white')
  File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\tkinter\__init__.py", line 2891, in create_rectangle
    return self._create('rectangle', args, kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\tkinter\__init__.py", line 2861, in _create
    return self.tk.getint(self.tk.call(
                          ^^^^^^^^^^^^^
_tkinter.TclError: unknown option "-bg"_'

The tape gets read though and the produced tap files do work.

[image]

InsaneDruid, 2024-09-05:
I don't pretent to be a python coder, but changing the contentWindow.update_idletasks() function calls in the tapeRead() function to contentWindow.update() ones solved the issue, now the window responds with "press play on tape" -> "searching" -> "reading from tape" and showing the read data, too!

wagiminator, 2024-09-07:
Hi, thanks a lot for the information. I’m not a Python coder myself; I somehow taught myself the little bit I needed ;-) If it works now, feel free to submit a pull request. I assume that it will work for me as well with the change.

wagiminator, 2024-09-08:
Already made the fix.

InsaneDruid, 2024-09-08:
Oh cool. I was just experimenting with my one and discovered the same issue on the progress bar for writing, i see you fixed that too. Nice!

Thanks

closed 2024-09-08 by InsaneDruid